benirose/O_C-BenisphereSuite #5: Shredder crash

razvysme, 2021-12-31:
If I hold the left encoder while Shredder is loaded and selected, it leads to overwriting the T1 with noise(random values, super fast). Only solution is to reset from menu, or reboot. I can replicate it if necessary.

P.S. Awesome of you to continue developing applets.

razvysme, 2022-01-17:
Sure, below is the link to the clip. Clock comes from Pamela, and Out A goes into Piston Honda 1V/Oct, nothing more. You can see that once I release the the right encoder after holding for a second or so, things go haywire. This behavior persist for other machines as well, just as I would expect it to behave if I'd insert noise into CV 1 (unfortunately I don't have a noise source).  https://photos.app.goo.gl/gA85YSRYa8x5q8EL9

herrkami, 2022-02-02:
Hey razvysme, 
As holding the left encoder enables the internal clock, could it be that what you see is just the clock going on? If your internal clock is configured to a very high speed or a high multiplier, i would expect such behavior. You can check your clock settings when you long-hold the down control button.

benirose, 2022-02-08:
Sorry for taking so long to circle back on this! I'll take a look into the issue this week and either work on a fix or close this out.

benirose, 2022-02-08:
Also, to get out of internal clock mode, you can change it to "forward" and then when you're back on your app screens, hold down the left encoder to disable clock forwarding.

benirose, 2022-02-22:
I'm going to close this out, if you see any problems again, feel free to open a new issue or comment on this one.

closed 2022-02-22 by benirose