benirose/O_C-BenisphereSuite #6: DrumMap questions: Expected PPQN? Interest in a state saving PR?

qiemem, 2022-01-12:
Loving this firmware so far!

Two questions:

1) I know grids can use either 4, 8, or 24 ppqn. What does DrumMap expect? From playing around, it seems like it expects 8 ppqn, but just wanted to confirm. Also, might be worth a mention in the manual as 4 ppqn is so common in eurorack.

2) I made a little fork of DrumMap that saves the state of all parameters, but at the cost of losing some granularity for manual control (cv control still has 8 bits of resolution). The resolutions I've used are: 6 bits each for x and y, 5 bits each for fills, 4 bits for chaos. With 2 bits for the modes and 2 for the cv mode, that's 32. Now, it's very much a trade-off, so I definitely understand if you don't want it. Also, it means that there's not quite enough values for the input acceleration to still be useful (which made me a bit sad tbh; that's such a great touch). Anyway, let me know!

benirose, 2022-01-14:
Thanks for checking it out!

1. Yes, the clock expects 8ppqn (32nd notes) as that's how the drum maps are programmed and this app simply advances one step per clock. I've had fun giving it 4ppqn for a half time beat too! I will add that to the wiki, thanks!
2. Hmm, at first I thought you were saying you scaled the values before saving and then rescaled them when loading, but now I'm realizing you changed them outright. I wonder if just scaling on save and load could work, I'm not sure if that's worth the trade off of approximate serialization though. Really chaos has no need to be 8-bits, but for the others, each value makes a difference. Let me think on this.

qiemem, 2022-01-14:
> Yes, the clock expects 8ppqn 

Great, thanks for confirming! The Grids manual mentions 24ppqn and triplets, but it wasn't clear to me how those could be represented in the 32 steps in the tables, so I was a bit confused.

> I wonder if just scaling on save and load could work,

I did this initially (since it's a trivial change) but wasn't crazy about how it wasn't clear how things would change on restart. The way I did it, there's the manually entered, lower granularity version that is re-scaled using `value * 255 / ((1 << bits) - 1)` (so you get the full 0 through 255 range). CV  then operates on that 8 bit value.

I made a quick table from the lut generator in the Grids repo to figure out what a good split would be. Numbers are the total number of a possible patterns for one channel given instrument, x, y, and fill (bit values in first row/col, number I picked bolded):

|xy/fill|3	|4	|5	|6	|7	|8     	|
|---   	|---    |---    |---    |---    |---    |---    |
| 3	|1014	|1573	|1990	|2224	|2345	|2422	|
| 4	|3290	|5002	|6272	|7015	|7483	|7702	|
| 5	|8603	|12403	|15016	|16653	|17623	|18136	|
| 6	|16412	|22553	|**26480**|28786	|30203	|30942	|
| 7	|23691	|32057	|36800	|39484	|41078	|42010	|
| 8	|28775	|38582	|43797	|46543	|48121	|48953	|

Worth noting this slightly underweights the two fills, as it only considers one channel; the 7-4 split felt too coarse for fill, so went with 6-5 even though it there's fewer patterns for a given channel.

Considering I lopped off 10 bits of information, keeping over half the patterns available felt okay, especially since patterns in the vicinity of the missing patterns will be available. The way I thought about it, compared to original Grids, I'm not sure I could reliably dial in more than 5 or 6 bits of information on an analog pot anyway. 

Hmm, now that I think about it... there's less than 16 bits of patterns... so I suppose it could use some clever remapping of encoder-entered-values to hit every possible pattern (but then leave response to CV unchanged)... sounds tricky but I could play around with it I suppose.

benirose, 2022-03-23:
This will be addressed in the 1.2 release.

closed 2022-03-23 by benirose