benirose/O_C-BenisphereSuite #18: LowerRenz misbehaviour

bowlneudel, 2022-03-22:
LowerRenz applet appears to have some bugs vs Hemispheres1.8A

Issue 1: Right hemisphere locked to left — adjusting settings on the right hemisphere does not have any effect. When LowerRenz is loaded on the left, it responds to Freq and Rho adjustment (which apply to the right hemisphere even when the left hemisphere is occupied by another applet).

Issue 2: Within one hemisphere, X & Y outputs are identical, instead of their expected pseudo-independent behaviour. In combination with Issue 1, all 4 outputs are identical.

bowlneudel, 2022-03-22:
[image]

benirose, 2022-03-30:
Hmm, I don't believe I modified this code at all but perhaps there were some other changes that impacted it.

bowlneudel, 2022-05-27:
Maybe it was a bug on my end, but it's fixed in 1.2! Thanks!

closed 2022-05-27 by bowlneudel

benirose, 2022-05-28:
Glad to hear this worked itself out too!