benirose/O_C-BenisphereSuite #19: Internal clock mode misbehaviour

bowlneudel, 2022-03-22:
Issue 1: When o_C is saved in external clock mode & forwarding off, Benispheres always boots with forwarding on after a power cycle.

Issue 2: Long-pressing the left encoder to toggle forwarding / metronome is inconsistent when certain applets are loaded (specifically LowerRenz, maybe others). The toggle may be completely unresponsive after a power cycle, until the clock mode is cycled or applets are changed.

benirose, 2022-03-23:
I've had a similar issue before and it may be related to some weirdness in EEPROM. It eventually went away, but I was never able to nail it down. Part of the issue was that the value being loaded for division or tempo of the internal clock was invalid (0?) and fixing that allowed me to disable the internal clock.

I'll see if I can reproduce any of this.

bowlneudel, 2022-05-27:
1.2 solved this for me!

closed 2022-05-27 by bowlneudel

benirose, 2022-05-28:
Glad to hear! I didn't address any of the clocking code, which just further leads me to believe that it was an EEPROM issue since I often see my settings get cleared up when I'm swapping out versions for testing.