benirose/O_C-BenisphereSuite #22: Stairs/stepper don't run in the flipped version of the firmware

glibglobber, 2022-03-27:
I switched over to unflipped and they run just fine, though.

benirose, 2022-10-08:
Is this still an issue in the latest release?

spacelordmother, 2022-12-03:
I'm still having this issue on version 1.3.1.   Also with apps Carpeggio and Shredder.

djphazer, 2023-02-12:
After some testing, I believe the root of this issue is that calibration data is improperly applied to the CV inputs in the flipped firmware. All 4 inputs on my module end up with a slight offset, which would cause some applets to appear unresponsive. Still not quite sure of the simplest proper solution tho...

djphazer, 2023-02-17:
Fix - https://github.com/djphazer/O_C-BenisphereSuite/commit/0bd6a6f82e3f0137df40c0796547019acae4d8fc

Not as elegant as I hoped, but it works. A better approach might be to reverse the arrays of calibration data on load (and save)