benirose/O_C-BenisphereSuite #23: Internal clock overrides external gates

glibglobber, 2022-03-27:
I'm trying to use Benispheres as a self-contained sequencer. For example: 

Internal clock > left side running a trigger sequencer > right side running a pitch/cv sequencer (such as stepper, tb-3P0, etc.). 

But I can't clock the pitch sequencer using the gates/triggers from the trigger sequencer. The internal clock replaces whatever clock I send to the Digital Ins. This means I have to use an external clock source in order to use a trigger and pitch sequencer.

benirose, 2022-03-30:
This is correct, right now the behavior of the internal clock is to feed both applets and replace the physical clock input on applets that have it. It is likely possible to add a new setting to the clock configuration screen that allows you to choose Left, Right, or Both hemispheres when using Internal clock. I'll put this in the backlog as an enhancement for a future release, thanks for the idea!

djphazer, 2022-11-14:
I recently hit this limitation after separating my O_C from Pamela. I just want the physical clock input to override the internal clock, mainly on the right side. I'll be looking for a solution...

benirose, 2022-11-14:
@djphazer if you want to take a swing at adding a setting to the Internal clock app for which side (left, right, both) to send the internal clock to, I would welcome that PR. Even if you just get the setting in the app but it doesn't function that could help. There's other code throughout that will need to be updated for getting the internal clock to respect that setting, primarily in `Hemipshere::Clock`. It's possible we can just change `clock_m->IsRunning()` to `clock_m->IsRunning(hemisphere)` and then in `ClockManager` have it cross reference that new setting.