benirose/O_C-BenisphereSuite #29: Feature request: AnnularFusion CV remap to Hits

bowlneudel, 2022-06-04:
Personally, I think it'd be more useful to have modulation control over the pattern fill versus the pattern offset, especially in combination with a shift register like Mylar Melodies demonstrates here: https://www.youtube.com/watch?v=d8L4NPg8gh0

Although, for backwards compatibility, this would probably have to be assigned via the encoder. Perhaps with the option to assign both CV parameters to Ch1 (controlling both steps and offset), and leave Ch2 for more static use, as described below:

| Encoder position | CV 1 | CV 2|
| --- | ---- | --- |
| 1 | Ch1 Rotate | Ch2 Rotate |
| 2 | Ch1 Hits | Ch2 Hits |
| 3 | Ch1 Hits | Ch1 Rotate |

benirose, 2022-10-08:
The reason CV controls rotate is because there's no UI on the screen for it. I'm considering pulling in [this version](https://modwiggler.com/forum/viewtopic.php?p=3677485#p3677485) of AnnularFusion. It gives control over all three parameters without having to use CV. In this case I would be fairly comfortable changing the CV inputs to steps. If I can find a way to make it so you can assign easily I might go down that path.

djphazer, 2022-10-09:
That updated version from qiemem looks promising! I've been using a different implementation from adegani called Euclid, where you can assign each CV input to any of the 6 parameters, but it doesn't have the visual representation of the pattern... I might attempt to cook the two together.

benirose, 2022-10-09:
I just read about Adegani branch on the modwiggler thread, looks like they took a few things from here, I might "borrow" the Random Walk applet! I'll take a look at their Euclid applet and see if we can't come up with a best of of both worlds situation!

benirose, 2022-10-09:
Oh weird, it's not in the release just in the source code, I'll have to pull it in and rebuild to check it out.