benirose/O_C-BenisphereSuite #33: Fork New Hemisphere 1.99

Bridgee, 2022-07-17:
Hemisphere 1.99 fixed number for bugs for the existing applets. I think it would be great if the benisphere also update that part.

yratof, 2022-07-20:
Yeah, I'm currently just porting over the changes from the new Hem, any chance you'll be upstreaming @benirose ?

benirose, 2022-07-20:
Yes, I'd like to! I have to look through the fixes. I will probably bring most of them in, but the changes to sequencer resets I'm not sure if it was implemented in the way I'm hoping to fix them in 1.3. If someone wants to put a PR together with the changes I can start looking through it and commenting.

People have been chomping for VOR support, maybe I'll pivot from the applets I'm still wrapping up and focus on that and upstreaming for 1.3 and then put the rest of the things I planned for 1.3 in 1.4 just so I can get a release out sooner.

Also, I'm on the fence about brining in Button as a replacement applet for Binary. Let me know your thoughts on that, if I can I'll make space for both of them, but I'd hate to delete a unique applet like Binary to make room for a utility applet like Button.

djphazer, 2022-07-21:
It seems like Binary was just too niche and not very useful; I never use it and won't miss it.
Though mundane, I appreciate the utility of Button.
my 2 cents

benirose, 2022-07-21:
In 1.99, there are 4 sequencer applets that were updated so that if reset is held high, clocks are ignored. This was done in an attempt to address issues where sending a reset at the same time you send a clock resulted in jumping to step 2. I don't feel this is correct behavior, especially if you're using a clock divider for reset, which will stay high for half the cycle.

I'm wondering if just changing them from `if (Clock(0) && !Gate(1))` to `if (Clock(0) && !Clock(1))` is good enough, this will mean only when clock and reset are triggered at the same time will it ignore the clock, and if reset is still high on the next clock it will continue as normal.

yratof, 2022-07-21:
If clock is 0 and is not 1? That will be only when clock is zero regardless of the reset?

benirose, 2022-07-21:
No sorry, `Clock(0)` returns true if there's a trigger on Digital input 1/3 (depending on the hemisphere the applet is running in) and likewise `Clock(1)` returns true if there's a trigger on Digital input 2/4. The parameter is for which input, not if it's low or high.

closed 2022-10-08 by benirose

benirose, 2022-10-09:
I was able to address the sequencing issue by having reset flags which prevent the sequencer from advancing while still triggering the first step. This allows sequencers to reset to step 1 while still playing step 1 regardless if clock is received at the same time or a later time.

yratof, 2022-10-09:
Lovely stuff, let’s get it merges

benirose, 2022-10-09:
It's merged! It's in the production branch as well as BSS1.3, which is ready for beta. I'm working on a write up to post it up.

yratof, 2022-10-09:
Awesome, thanks for the update!

PR #36 (2022-10-08): Merge upstream
Hemispheres recently released a final version of 1.99repeating, and there were several bug fixes and even a new applet to replace an old one! I'm probably not going to do a blind merge in, as I know I have opinions on how the reset changes were done, and I'm not sure if I'm willing to lose Binary for Button yet.

Closes #33