benirose/O_C-BenisphereSuite #34: Change reset behavior on sequencers - when NOT clocked and reset, next clock should play first step

benirose, 2022-07-18:
The current behavior for reset in most applets is to set things back to the first step. This is a pretty analog approach, but is probably not the expected behavior, as the next clock will move it to the second step rather than playing the first one. If possible, applets should be updated so that if they are reset while not receiving a clock, the first clock they receive plays the first step.

benirose, 2022-10-08:
This has all been address in BSS1.3 branch.

closed 2022-10-08 by benirose

djphazer, 2022-10-15:
Looking thru your new reset code tweaks, I can't help but feel like they're a little over-engineered... I'm sure it works as intended, haven't fully tested myself.

The simpler solution in my mind is to process and output the current step on Clock(0) _first_ and increment the step _last_. So, if a reset sets it back to step=0, that step is still processed before advancing. This is how it already works in AnnularFusion.