benirose/O_C-BenisphereSuite #37: State Not Saving Via Right Hold

carey-james, 2022-07-30:
I am having trouble saving the state of the unit/start up apps. I assumed the way to do so would be the same as base Hemispheres, by holding the Right button for two long presses, but after opening the internal clock it does not seem like the state is saving. Not sure if this is a bug or the feature has been removed from Hemispheres.

benirose, 2022-09-01:
Hey there. You can definitely save state, but I can't remember if internal clock settings are included or not. If not, I should definitely get them in there!

carey-james, 2022-09-02:
I have to apologize! I've discovered this is a hardware issue with my encoder that just happened develop around the same time as I installed Benispheres! I've repaired and it is now behaving as expected.

closed 2022-09-02 by carey-james