benirose/O_C-BenisphereSuite #38: Compile error when swapping Enigma out for another app in OC_options.h

nickstokes, 2022-08-09:
When I swapped out ENABLE_APP_ENIGMA for another app in OC_options.h I get the compile error **'OC::Strings' has not been declared**

It looks like OC_strings.h is only included in APP_ENIGMA.ino, but is needed for OC_calibration.ino to compile.
Moving the line `#include "OC_strings.h"` from APP_ENIGMA.ino to o_c_REV.ino seems to fix it.

benirose, 2022-08-10:
Yeah I need to find the root of it. My hunch is that there's stuff in EnigmaJr that expects Enigma to be there. Try removing that app from `hemisphere_config.h` and see if it works. You might have to change `HEM_EnigmaJr.ino` to something like `HEM_EnigmaJr.ino.disabled` because the compiler is a bit greedy if it sees .ino files. Alternatively you can wrap all of the code in `HEM_EnigmaJr.ino` in `IFDEF ENABLE_APP_ENIGMA` so it's driven by the same flag. However more work needs to be done to get this flag to remove it from `hemisphere_config.h`, since we're still hardcoding the applet count (which I also need to address).

FWIW Enigma and EnigmaJr are on the chopping block because I'm running out of space again. Curious what you're doing for your build chain because I know it's a pain with the legacy 32-bit versions and what not. I'd like to get this containerized, especially so we can then make a web tool to compile your own version.

nickstokes, 2022-08-14:
Just moving the `#include "OC_string.h"` line from APP_ENIGMA.ino to o_c_REV.ino is all I needed to do to fix it.
Strangely enough, EnigmaJr still works fine, and swapping out ENABLE_APP_ENGIMA freed up enough space to include another app like Captain MIDI.

commit 3fbb076d (2023-02-09): Add OC_strings to DualQuant in case enigma app is disabled. Fix #38
commit ed3910d5 (2023-02-09): Merge pull request #56 from Nagasaki45/no-enigma

Add OC_strings to DualQuant in case enigma app is disabled. Fix #38
closed 2023-02-09 by benirose