benirose/O_C-BenisphereSuite #39: Scale duet not working?

risro, 2022-08-20:
am I the only one who has an issue with scale duet? Can't seem to get anything out of it no matter what I try.

Heres what I have connected at the moment:
MATHS channel 1 cycling sending a 0-5v sine to O&C cv3
MATHS channel 4 cycling and EOC to O&C trig 3
O&C OUT C to oscilloscope and it's not showing anything (also same with OUT D)
doens't matter if I have selected all the keys or some keys on scale duet.
if I send MATHS channel 1 to oscilloscope I see the 0-5v sine wave no problems.

am I just not understanding how it should work?

yratof, 2022-08-23:
Are you also triggering scale duet? CV + Trigger are needed to quantise

risro, 2022-08-24:
yes the maths EOC (end of cycle) to o&c trig 1 (or 3)  should do that right? i also tried other triggers, but nothing seems to happen. 

i also tried the latest hemisphere wirmware and couldn't get the quantizer to work there eighter so I wonder if my o&c is broken somehow. everything else seems to work though 🤔

yratof, 2022-08-24:
Have you tried sending a regular trigger and just an lfo, I have a feeling maths might not be triggering with EOC, maybe it’s more gate than trigger to o_c

risro, 2022-08-24:
i did also try with varigate 4+ and a Turing machine so I don't think that's the problem. for some reason it just doesn't seem to do anything no matter what I try.

benirose, 2022-09-16:
@yratof Thanks for helping out on this one! @risro Feel free to reopen if you have any other issues. I've definitely noticed a pattern of reliability issues that seem random and hard to replicate, so I'm hoping to spend some time in the future just digging into those.

closed 2022-09-16 by benirose