benirose/O_C-BenisphereSuite #40: Scroll backwards through parameters

khogeland, 2022-08-30:
It would be really nice if you could scroll through parameters by holding down the encoder and rotating back/forth, Disting-style. I appreciate the simplicity of the UI, but some apps (Scale Duet, Shredder) are pretty annoying to navigate in only one direction. Curious to see what folks think- I'm not sure if making it easier to navigate might counter-intuitively degrade the experience by encouraging more complex apps... y'know, Disting-style.

djphazer, 2022-08-31:
This would probably take some work, but I do like the idea of rotating to select and then pushing to toggle editing of each parameter. Beni's new ProbMelo applet works like this.

djphazer, 2022-09-01:
I just implemented this for ShiftReg and TB-3PO on my branch, and I like it. It makes the code/binary slightly larger tho, so I had to disable other applets to make it fit. Maybe some shared functions could mitigate this, idk.

To be clear tho, it's push to toggle editing... not the Disting-style hold-and-rotate as suggested. That seems too complicated for me!

khogeland, 2022-09-01:
The hold-and-rotate wouldn't mess with peoples' current muscle memory, plus, there's less chance of a mistake. With a toggle, if you accidentally press (or don't press) the encoder, you're going to have a confusing jolt when you scroll the wrong thing and have to backtrack, especially if you go interact with something else in your setup and come back. Looks like it would require changes to how UI state is communicated to the app though (maybe just a button release event? I only looked briefly).

khogeland, 2022-09-01:
The hold-and-rotate wouldn't mess with peoples' current muscle memory, plus, there's less chance of a mistake. With a toggle, if you accidentally press (or don't press) the encoder, you're going to have a confusing jolt when you scroll the wrong thing and have to backtrack, especially if you go interact with something else in your setup and come back. It's already frustrating when I scroll the wrong parameter, now double that 😆  Looks like it would require changes to how UI state is communicated to the app though (maybe just a button release event? I only looked briefly).

benirose, 2022-09-16:
I understand the thought here and it has rattled around in my mind as well, especially an app like Shredder where I had to compromise with responsiveness in order to implement double-press for shred. However, I think one of the guiding principles of Hemispheres is that applets should be simple enough with few enough parameters that cycling through them should be easy. There are a few applets where this isn't possible, namely ScaleDuet and my upcoming ProbMelo (which took the UI directly from ScaleDuet) where modal editing is needed (push to change between parameters selection and parameter editing), but ideally if an applet has too many parameters to cycle it isn't right for Hemispheres. I was even a bit hesitant on BugCrack for this reason but in the end I think it just barely passed the test. 

With all of this said, I am open to experimenting in a separate branch and trying out different input mechanism that could make navigating easier. I wonder what having all apps use modal editing would be like.

djphazer, 2022-09-18:
I really enjoy the modal editing style, feels faster and more intuitive to me. Curious to see if other people would agree. I'll gladly contribute to a separate branch; I was already thinking of adapting the rest of the applets for my own use.