benirose/O_C-BenisphereSuite #47: Scroll direction of right knob reversed?

5-lagu, 2022-10-31:
I just updated to v1.3 and now the right encoder on my uO_C has reversed scrolling direction, while the left is normal. Is anyone else experiencing this? I haven't tried reverting to another firmware (yet) to confirm that it only happened with Benisphere v1.3...

djphazer, 2022-10-31:
This is a calibration setting... https://ornament-and-cri.me/calibration/

I was just as confused as you until I RTFM

5-lagu, 2022-11-01:
Ok, that was a weird one... esp. "no visual feedback" which is quite unhelpful in a calibration setting...

benirose, 2022-11-28:
Yeah I agree, I used to have mine flipped and hating having no idea what it was set to! I'll open a separate ticket to change that and put it in the backlog!

closed 2022-11-28 by benirose