benirose/O_C-BenisphereSuite #62: Feature Request: consider extending Sequence5 into having more steps

gilbertohasnofb, 2023-01-07:
Sequence5 is a brilliant little app for when you need a small sequencer, but I wonder if it would be possible to extend it to a more traditional 8 steps. Now that you've implemented the reset to all sequencers and given that you can ignore a step by setting its value to 0, that would make this a really powerful sequencing tool. UI-wise, I think that 8 sliders could still be comfortably displayed in half screen, as is used by your ProbMeloD app.

djphazer, 2023-01-07:
I've already done this in my branch!
[Link to commit](https://github.com/djphazer/O_C-BenisphereSuite/commit/21de359170a3d0877a0bd851a977b3790e585d0d)

gilbertohasnofb, 2023-01-07:
@djphazer That's absolutely brilliant, thank you so much for this -- I'll have a look at your fork! Two small things to ask you:

* I'm looking at the code of your file `HEM_Sequence5.ino` in your repo; have you considered renaming both the source code `.ino` file and the app itself (line 26) from `Sequence5` to `Sequence8` or  `SequenceX` or even just `Sequencer`?
* I see that you don't build `.hex` files in your repo -- would you considering opening a PR over here if @benirose would be interested in this implementation?

djphazer, 2023-01-07:
Renaming the file is nbd, I just left it as is for historical reasons, so it's easy to see the diff against other forks.

Beni has already expressed interest in pulling in a lot of my work, but he's been busy... I'm planning on releasing a .hex file soon! Meanwhile, it's pretty easy to build it yourself. Check the README on my fork.

gilbertohasnofb, 2023-01-07:
@djphazer great, thanks for the info! I'll try to build it myself when I have a chance.