benirose/O_C-BenisphereSuite #63: VOR support

asher, 2023-01-17:
Saw in forums that this is already planned, but opening an issue to track. I recently picked up an OCP X and would be happy to help with testing or with resolving issues if you've got something in progress in a local branch.

djphazer, 2023-01-17:
Excellent! I have a [proof-of-concept for you to test](https://github.com/djphazer/O_C-BenisphereSuite/releases/tag/PSv1.4.9%2BVOR)

This is the result of merging my **phazerville** branch into the upstream **ocplus** branch from Chysn. I'll try to rebase just the VOR stuff as a patch on Beni's **production** branch. [Source tree is here](https://github.com/djphazer/O_C-BenisphereSuite/tree/ocplus)

djphazer, 2023-01-17:
Isolated patch against production: https://github.com/djphazer/O_C-BenisphereSuite/tree/vor-patch

asher, 2023-01-18:
Thanks for this!! Just installed your phazerville build and so far have noticed that the VOR button is inactive, though long pressing up (enabled via `#define VOR_NO_RANGE_BUTTON`) does open the range selection modal. The only app I've tried so far is AD Env. In -5/5v mode, it produces 0-5v envs as with a standard build. The other ranges just act as an offset though, in -3/7v mode, the 0 point becomes a constant 2v resulting in 2-7v envelopes, and in 0/10v mode, the zero point becomes 5v, with 5-10v envs. 

I won't have time before the weekend, but I'm looking forward to playing around with the `vor-patch` branch and will see if I can make some progress with this.

djphazer, 2023-02-08:
Any progress or insights on this? We're trying to tackle VOR support today...

I will say that many applets are probably designed with naive assumptions about the output voltage range, but it should be relatively easy to tweak them for proper expected behavior.

opeologist, 2023-04-21:
i have a 1UO_C that i believe needs the VOR version of this, but don't have a legacy MacOS device to build from source with. is there any way to link a built hex to flash with? thanks!

djphazer, 2023-04-21:
There's a VOR build available on [my fork](https://github.com/djphazer/O_C-BenisphereSuite/releases/tag/PSv1.5.1)