benirose/O_C-BenisphereSuite #64: Probdiv reseed CV affects Probmelod lower bound range

rheslip, 2023-01-19:
Suite V 1.3.1. using Probdiv on the left and Probmelod on the right. Using an LFO CV to reseed Probdiv every few seconds. It works for a few cycles and then Probmelod lower range gets set to the upper range after which all the notes remain the same. Seems like CV2 gets interpreted as CV3.

rheslip, 2023-01-19:
Also noticed that Probdiv loop length gets reset to off randomly with the above setup. I've reproduced both issues on a second O-C so its not the hardware. Somehow modulating CV2 randomly behaves as if its modulating CV1 and CV3. If I don't modulate CV2 everything behaves as it should.

djphazer, 2023-01-26:
I'm getting the same weird behavior... haven't really tested these two together that much yet.

I don't have an explanation for this bug, but one thing that bothers me is the CV inputs overwrite the settings (loop length, lower bound, upper bound) rather than offset them. I might have to rewrite some of this logic...

cmovvv, 2023-10-06:
I'm running into this as well, but only when reseeding ProbDiv - otherwise, the range and loop stays static as desired. 

Any developments on addressing this in a future update? When the reseed function works, it's a really great way of making generative melodies on the O_C.

djphazer, 2023-10-06:
This might be related to a bug where the ADC inputs can bleed into each other. It should be fixed in Phazerville v1.6.x