benirose/O_C-BenisphereSuite #69: Vector LFO - modification request

X-modular, 2023-02-14:
Vector LFO is currently LFO1 and LFO2 that can have independent shapes and rates.  LFO1 is always discreet in it's channel.  However, LFO2 currently defaults to a 50% blend of LFO1 and LFO2.  The blend can be reduced to zero LFO1 and pure LFO2 by applying a CV input.  This requires extra modules just to have two truly independent LFOs.  

I think most people would want the default behavior to be pure LFO2 on it's channel.  Then a CV could be applied to achieve a blend of LFOs on the LFO2 channel as a special effect rather than the default.  

The blend could be 0-100% LFO1 plus 100% of LFO2.  

A cross fade mode is another possibility.  LFO2 decreases linearly as LFO1 increases.

djphazer, 2023-02-17:
I agree - https://github.com/djphazer/O_C-BenisphereSuite/commit/680a51cc61911189b17184026b384fca266d8d40

Plus I made the CV input bipolar. I kept it as a balanced 50/50 blend at max, but I might still try out the full crossfade...

X-modular, 2023-02-17:
Nice.  That will work for me.  The bipolar input is a good addition.  Thanks.